probe-rs/embedded-test #15: Allow to run host side code (rust?/bash?) before/after each test case

t-moe, 2023-12-22:
I'm open to take design ideas...

Running a bash script could be pretty doable. Running rust code, would require to build it somehow and then load it as library into probe-rs?

t-moe, 2023-12-28:
As a very basic first approach, we could even store the bash script on the target and transfer it together with the list of tests. 

But yes... It would be really cool if would be able to mix host and target side rust code. We could auto-insert breakpoints on the target, to switch from the target executing to host....

Yatekii, 2023-12-28:
The problem is how you configure the compilation (pull in crates). Maybe it could be a hack via cargo metadata where you can define deps like that. but r-a etc will not work :/

ejpcmac, 2024-07-09:
Another use-case for running host-side code in the test runner would be to drive a counterpart. Like sending / receiving data via UART using a FTDI to test the UART HAL on the chip, or driving another MCU to check for GPIO states, and so on.

I was thinking recently how one could achieve that with linear-looking code, and had the idea of writing embed fragments in the code, like (pseudocode):

```rust
#[test]
fn test_gpio() {
    let counterpart = Counterpart::init();
    assert!(counterpart.gpio0.is_low());

    embedded! {
        let dp = Peripherals::take();
        let gpio0 = dp.GPIO0;
        gpio0.set_high();
    }

    assert!(counterpart.gpio0.is_high());
}
```

When compiling, all `embedded!` fragments would be extracted to be compiled for the target architecture in the test firmware, and replaced to an RPC call to the embedded firmware in the test runner.

So, from the test writer point of view, the test would be run from the host, except `embedded!` fragments which would be run on the embedded system. If status / data is needed, one could write:

```rust
let output = embedded! {
    let value_computed_on_embedded_side = some_computation();
    value_computed_on_embedded_side
};
```