probe-rs/embedded-test #18: backtrace is not helpful on panic

t-moe, 2023-12-28:
@Vollbrecht reported the following output, printed by `probe-rs test` on assertion fail:

```
panicked at tests/example_test.rs:173:9:
assertion failed: false
Frame 0: syscall_readonly @ 0x42002124 inline
       /home/fred/.cargo/registry/src/index.crates.io-6f17d22bba15001f/semihosting-0.1.5/src/sys/arm_compat/syscall/riscv.rs:35:9
Frame 1: sys_exit @ 0x0000000042002112 inline
       /home/fred/.cargo/registry/src/index.crates.io-6f17d22bba15001f/semihosting-0.1.5/src/sys/arm_compat/mod.rs:183:9
Frame 2: exit @ 0x0000000042002112 inline
       /home/fred/.cargo/registry/src/index.crates.io-6f17d22bba15001f/semihosting-0.1.5/src/sys/arm_compat/mod.rs:172:5
Frame 3: exit @ 0x00000000420020ec
       /home/fred/.cargo/registry/src/index.crates.io-6f17d22bba15001f/semihosting-0.1.5/src/process.rs:14:5
Frame 4: <unknown function @ 0x4200213a> @ 0x4200213a
```

The backtrace contains only internals (frame 0 - 3), but no frames from user code, so it is not really helpful.

bugadani, 2024-11-08:
This is not an embedded-test issue.

closed 2024-11-08 by bugadani

bugadani, 2025-09-17:
probe-rs should be able to unwind backtraces, but there are a few different issues with the different architectures. This particular issue is probably https://github.com/probe-rs/probe-rs/issues/3408