probe-rs/embedded-test #19: defmt dependency conflics

Artur-Romaniuk, 2024-01-09:
In order to use defmt logging, one has to enable "defmt" feature for embedded-test.
```
embedded-test = { version = "0.2.2", features = [
    "defmt",
    "embassy",
] }
```
This resolved to the internal dependency of embedded-test: defmt v0.3.5

At the same time, docs say that one should use patched panic-probe
```
panic-probe = { git = "https://github.com/t-moe/defmt", features = [
    "print-defmt",
] }
```
and replace all other uses of defmt with patched version:
```
defmt = { git = "https://github.com/t-moe/defmt" }
defmt-rtt = { git = "https://github.com/t-moe/defmt" }
```

This results in a dependency conflict like this:
```
    Updating git repository `https://github.com/t-moe/defmt`
error: failed to select a version for `defmt`.
    ... required by package `embedded-test v0.2.2`
    ... which satisfies dependency `embedded-test = "^0.2.2"` (locked to 0.2.2) of package `target-tests v0.1.0 (/home/aromaniuk/Repos/qca7000-rs/target-tests)`
versions that meet the requirements `^0.3.5` (locked to 0.3.5) are: 0.3.5

the package `defmt` links to the native library `defmt`, but it conflicts with a previous package which links to `defmt` as well:
package `defmt v0.3.5 (https://github.com/t-moe/defmt#7c4c12d6)`
    ... which satisfies git dependency `defmt` of package `target-tests v0.1.0 (/home/aromaniuk/Repos/qca7000-rs/target-tests)`
```

Maybe docs should be updated to better describe how to use "print-defmt" feature, as all examples have only "print-log",

t-moe, 2024-01-09:
Hello and thank you for that report.

try to add

```
[patch.crates-io]
defmt = { git = "https://github.com/t-moe/defmt" }
defmt-rtt = { git = "https://github.com/t-moe/defmt" }
```
so that the internal dependencies to panic-probe are also fixed.

I'm still hoping that the defmt team will at some point merge my PR, currently they seem to be busy with other stuff.

I'll consider to fix the docs, if this takes longer ;)

t-moe, 2024-01-09:
If updated the readme in this repo and the cargo.toml here and in the example repo.

Hope this helps. Otherwise please report back.

closed 2024-01-09 by t-moe

Artur-Romaniuk, 2024-01-09:
Thank you. This is very helpful.
Especially, that I am currently debugging another issue and defmt revealed new info.