probe-rs/embedded-test #22: Can't compile with embassy feature

jclmnop, 2024-02-02:
Thought I'd give this crate a go because it has built-in support for async tests with embassy. 

However, I get a compile error even though I've enabled one of the `executor-thread`/`executor-interrupt` features on `embassy-executor`.

```toml
[dependencies]
embassy-embedded-hal = { version = "0.1.0", features = ["defmt"] }
embassy-sync = { version = "0.5.0", features = ["defmt"] }
embassy-executor = { version = "0.5.0", features = ["task-arena-size-32768", "arch-cortex-m", "executor-thread", "executor-interrupt", "defmt", "integrated-timers"] } # <--
embassy-time = { version = "0.3.0", features = ["defmt", "defmt-timestamp-uptime"] }
embassy-rp = { version = "0.1.0", features = ["defmt", "unstable-pac", "time-driver", "critical-section-impl"] }
embassy-futures = { version = "0.1.0" }

defmt = "0.3"
defmt-rtt = "0.4"
fixed = "1.23.1"
fixed-macro = "1.2"

#cortex-m = { version = "0.7.6", features = ["critical-section-single-core"] }
cortex-m = { version = "0.7.6", features = ["inline-asm"] }
cortex-m-rt = "0.7.0"
cortex-m-semihosting = "0.5"
panic-probe = { version = "0.3", features = ["print-defmt"] }
futures = { version = "0.3.17", default-features = false, features = ["async-await", "cfg-target-has-atomic", "unstable"] }
byte-slice-cast = { version = "1.2.0", default-features = false }
heapless = "0.8"

embedded-hal-1 = { package = "embedded-hal", version = "1.0" }
embedded-hal-async = "1.0"
embedded-hal-bus = { version = "0.1", features = ["async"] }
embedded-io-async = { version = "0.6.1", features = ["defmt-03"] }
embedded-storage = { version = "0.3" }
static_cell = "2"
portable-atomic = { version = "1.5", features = ["critical-section"] }
log = "0.4"
pio-proc = "0.2"
pio = "0.2.1"
rand = { version = "0.8.5", default-features = false }

byte = "0.2.6"

[dev-dependencies]
embedded-test = { version = "0.2.3", features = ["embassy", "defmt"] } # <--
```

The error:

```bash
error[E0432]: unresolved import `embassy_executor::Executor`
  --> /Users/jclmnop/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-test-0.2.3/src/export.rs:10:9
   |
10 | pub use embassy_executor::Executor; // Please activate the `executor-thread` or `executor-interrupt` feature on the embassy-executor crate!
   |         ^^^^^^^^^^^^^^^^^^^^^^^^^^ no `Executor` in the root
   |
help: consider importing this struct instead
   |
10 | pub use embassy_executor::raw::Executor; // Please activate the `executor-thread` or `executor-interrupt` feature on the embassy-executor crate!
   |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

Using an RP2040 with `embassy-rp`, not sure whether that's relevant or not though. Tried enabling just one of the executor features, disabling default features, etc, on `embassy-executor` but same issue. 

I don't expect any help or a quick fix for this, it's probably some sort of mistake on my end. Just dropping an issue here so you're aware of it, just in case it is something on your end.

t-moe, 2024-02-02:
This is probably because I'm still on embassy executor 0.4 and you're using embassy executor 0.5.

https://github.com/probe-rs/embedded-test/blob/0e836a690f04c4e8569cf86e5e4546837f7d590e/Cargo.toml#L22

t-moe, 2024-02-05:
I've published a new version v0.3.0, which uses embassy-executor v0.5.0. This should fix your problem. 
Please report back, if you still face any issues.

closed 2024-02-05 by t-moe