probe-rs/embedded-test #24: Runner does not install/compile

thedevleon, 2024-02-20:
I cloned the example repo and tried to run `cargo test`, which results in this error:

```
$ cargo test
    Finished test [unoptimized + debuginfo] target(s) in 0.03s
     Running tests/example_test.rs (target/riscv32imac-unknown-none-elf/debug/deps/example_test-ae604d781d4f9ebe)
error: unrecognized subcommand 'test'

  tip: a similar subcommand exists: 'reset'

Usage: probe-rs [OPTIONS] <COMMAND>

For more information, try '--help'.
error: test failed, to rerun pass `--test example_test`

Caused by:
  process didn't exit successfully: `probe-rs test --chip esp32c6 -- /home/leon/Desktop/ivy/code/embedded-test-example/target/riscv32imac-unknown-none-elf/debug/deps/example_test-ae604d781d4f9ebe` (exit status: 2)
```

Turns out, that the probe-rs runner was never installed with the test feature, because the following command fails to compile:
`cargo install probe-rs --git https://github.com/probe-rs/probe-rs --branch feature/testing --features cli --bin probe-rs` results in:

```
   Compiling capstone v0.11.0
error[E0603]: struct `Formatter` is private
   --> probe-rs/src/bin/probe-rs/util/rtt.rs:390:36
    |
390 |     formatter: defmt_decoder::log::Formatter,
    |                                    ^^^^^^^^^ private struct
    |
note: the struct `Formatter` is defined here
   --> /home/leon/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-decoder-0.3.10/src/log/mod.rs:19:14
    |
19  |     format::{Formatter, HostFormatter},
    |              ^^^^^^^^^
help: consider importing this struct instead
    |
390 |     formatter: defmt_decoder::log::format::Formatter,
    |                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
help: import `Formatter` directly
    |
390 |     formatter: defmt_decoder::log::format::Formatter,
    |                ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

error[E0603]: struct `Formatter` is private
   --> probe-rs/src/bin/probe-rs/util/rtt.rs:474:53
    |
474 |                 let formatter = defmt_decoder::log::Formatter::new(format);
    |                                                     ^^^^^^^^^ private struct
    |
note: the struct `Formatter` is defined here
   --> /home/leon/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-decoder-0.3.10/src/log/mod.rs:19:14
    |
19  |     format::{Formatter, HostFormatter},
    |              ^^^^^^^^^
help: consider importing this struct instead
    |
474 |                 let formatter = defmt_decoder::log::format::Formatter::new(format);
    |                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
help: import `Formatter` directly
    |
474 |                 let formatter = defmt_decoder::log::format::Formatter(format);
    |                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

error[E0599]: no method named `format_to_string` found for reference `&defmt_decoder::log::format::Formatter` in the current scope
   --> probe-rs/src/bin/probe-rs/util/rtt.rs:350:43
    |
350 | ...                   formatter.format_to_string(frame, file.as_deref(), line, module);
    |                                 ^^^^^^^^^^^^^^^^ method not found in `&Formatter`

error[E0277]: `defmt_decoder::log::format::Formatter` doesn't implement `std::fmt::Debug`
   --> probe-rs/src/bin/probe-rs/util/rtt.rs:390:5
    |
386 | #[derive(Debug)]
    |          ----- in this derive macro expansion
...
390 |     formatter: defmt_decoder::log::Formatter,
    |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ `defmt_decoder::log::format::Formatter` cannot be formatted using `{:?}` because it doesn't implement `std::fmt::Debug`
    |
    = help: the trait `std::fmt::Debug` is not implemented for `defmt_decoder::log::format::Formatter`
    = note: this error originates in the derive macro `Debug` (in Nightly builds, run with -Z macro-backtrace for more info)

error[E0308]: mismatched types
   --> probe-rs/src/bin/probe-rs/util/rtt.rs:474:68
    |
474 |                 let formatter = defmt_decoder::log::Formatter::new(format);
    |                                 ---------------------------------- ^^^^^^ expected `FormatterConfig<'_>`, found `&str`
    |                                 |
    |                                 arguments to this function are incorrect
    |
note: associated function defined here
   --> /home/leon/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-decoder-0.3.10/src/log/format/mod.rs:193:12
    |
193 |     pub fn new(config: FormatterConfig) -> Self {
    |            ^^^

Some errors have detailed explanations: E0277, E0308, E0599, E0603.
For more information about an error, try `rustc --explain E0277`.
error: could not compile `probe-rs` (bin "probe-rs") due to 5 previous errors
error: failed to compile `probe-rs v0.21.1 (https://github.com/probe-rs/probe-rs?branch=feature/testing#9d10d333)`, intermediate artifacts can be found at `/tmp/cargo-install9j2xhG`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.
```

thedevleon, 2024-02-20:
Installing with the testing2 branch works: `install probe-rs --git https://github.com/probe-rs/probe-rs --branch feature/testing2 --features cli --bin probe-rs`, but there are a bunch of errors when running:

```
$ cargo test
    Finished test [unoptimized + debuginfo] target(s) in 0.03s
     Running tests/example_test.rs (target/riscv32imac-unknown-none-elf/debug/deps/example_test-ae604d781d4f9ebe)
      Erasing ✔ [00:00:01] [###############################################################################################################################################################] 256.00 KiB/256.00 KiB @ 249.17 KiB/s (eta 0s )
  Programming ✔ [00:00:01] [##################################################################################################################################################################] 79.38 KiB/79.38 KiB @ 62.35 KiB/s (eta 0s )    Finished in 2.323s
 WARN probe_rs::rtt::syscall: Unknown semihosting operation=0015 parameter=4086e318
 WARN probe_rs::rtt::syscall: Unknown semihosting operation=0000 parameter=4086e318
 WARN probe_rs::rtt::syscall: Unknown semihosting operation=0100 parameter=4086fbd0

running 8 tests
test unit_tests::takes_state      ...  WARN probe_rs::rtt::syscall: Unknown semihosting operation=0015 parameter=4086e318
ERROR probe_rs::architecture::riscv::communication_interface: Executing the abstract command for write_32 failed: Exception (Abstractcs { .0: 10000302, progbufsize: 10, busy: false, cmderr: 3, datacount: 2 })
FAILED
test unit_tests::log              ...  WARN probe_rs::rtt::syscall: Unknown semihosting operation=0015 parameter=4086e318
ERROR probe_rs::architecture::riscv::communication_interface: Executing the abstract command for write_32 failed: Exception (Abstractcs { .0: 10000302, progbufsize: 10, busy: false, cmderr: 3, datacount: 2 })
FAILED
test unit_tests::it_works_ignored ... ignored
test unit_tests::it_fails1        ...  WARN probe_rs::rtt::syscall: Unknown semihosting operation=0015 parameter=4086e318
ERROR probe_rs::architecture::riscv::communication_interface: Executing the abstract command for write_32 failed: Exception (Abstractcs { .0: 10000302, progbufsize: 10, busy: false, cmderr: 3, datacount: 2 })
FAILED
test unit_tests::it_fails2        ...  WARN probe_rs::rtt::syscall: Unknown semihosting operation=0015 parameter=4086e318
ERROR probe_rs::architecture::riscv::communication_interface: Executing the abstract command for write_32 failed: Exception (Abstractcs { .0: 10000302, progbufsize: 10, busy: false, cmderr: 3, datacount: 2 })
FAILED
test unit_tests::it_passes        ...  WARN probe_rs::rtt::syscall: Unknown semihosting operation=0015 parameter=4086e318
ERROR probe_rs::architecture::riscv::communication_interface: Executing the abstract command for write_32 failed: Exception (Abstractcs { .0: 10000302, progbufsize: 10, busy: false, cmderr: 3, datacount: 2 })
FAILED
test unit_tests::it_fails3        ...  WARN probe_rs::rtt::syscall: Unknown semihosting operation=0015 parameter=4086e318
ERROR probe_rs::architecture::riscv::communication_interface: Executing the abstract command for write_32 failed: Exception (Abstractcs { .0: 10000302, progbufsize: 10, busy: false, cmderr: 3, datacount: 2 })
FAILED
test unit_tests::it_timeouts      ...  WARN probe_rs::rtt::syscall: Unknown semihosting operation=0015 parameter=4086e318
ERROR probe_rs::architecture::riscv::communication_interface: Executing the abstract command for write_32 failed: Exception (Abstractcs { .0: 10000302, progbufsize: 10, busy: false, cmderr: 3, datacount: 2 })
FAILED

failures:

---- unit_tests::takes_state ----
A RISC-V specific error occurred.

---- unit_tests::log ----
A RISC-V specific error occurred.

---- unit_tests::it_fails1 ----
A RISC-V specific error occurred.

---- unit_tests::it_fails2 ----
A RISC-V specific error occurred.

---- unit_tests::it_passes ----
A RISC-V specific error occurred.

---- unit_tests::it_fails3 ----
A RISC-V specific error occurred.

---- unit_tests::it_timeouts ----
A RISC-V specific error occurred.

failures:
    unit_tests::takes_state
    unit_tests::log
    unit_tests::it_fails1
    unit_tests::it_fails2
    unit_tests::it_passes
    unit_tests::it_fails3
    unit_tests::it_timeouts

test result: FAILED. 0 passed; 7 failed; 1 ignored; 0 measured; 0 filtered out; finished in 1.46s
```

t-moe, 2024-02-20:
Hei @thedevleon ,
Please use  add `--locked` for now to fix the defmt semver break:

`cargo install probe-rs --git https://github.com/probe-rs/probe-rs --branch feature/testing --features cli --bin probe-rs  --locked`

(The branch feature/testing2 is for development. Currently not stable)

t-moe, 2024-02-20:
EDIT:
I've fixed the `feature/testing` branch. `--locked` should no longer be required.

@thedevleon Out of interest: Which platform were you on when running the `feature/testing2` branch? https://github.com/probe-rs/embedded-test/issues/24#issuecomment-1954333228

Thank you

thedevleon, 2024-02-20:
> > I'm running the sample on the C6.
> 
> sorry I meant linux or windows. because we're having troubles with the testing2 branch on certain platforms.

Ah, I'm running Ubuntu 22.04.3.

closed 2024-02-20 by t-moe