probe-rs/embedded-test #25: Fails to run `cargo test` with existance of a `src/main.rs`

oshaughnessya, 2024-03-07:
Cargo Version: v1.78.0-nightly and 1.76.0-stable

probe-rs 0.22.0
git commit: 0f2eeb29

Reproducibility:
Using the embedded-test-example repository on the stm32f4 branch, add a main.rs, I am using 
```rs
#![deny(unsafe_code)]
#![no_main]
#![no_std]

use cortex_m_rt::entry;
use stm32f4xx_hal::{
    pac,
    prelude::*,
};

#[allow(clippy::empty_loop)]
#[entry]
fn main() -> ! {
    loop {}
}
```

after `src/main.rs` exists, run `cargo test`. You should get an error saying 
```
error[E0463]: can't find crate for `test`
```
Full output of `cargo test` is [here](https://github.com/probe-rs/embedded-test/files/14519077/error.txt).

t-moe, 2024-03-07:
Hmm, Apparently `cargo test` wants to also test the binaries... This is new to me, but is has probably been there for a while.

Adding the following to `Cargo.toml` helps as a workaround.

```
[[bin]]
name = "embedded-example-test"
path = "src/main.rs"
test = false
```

For me, running `cargo test` hardly makes sense anyways, because I usually have unit tests that will only compile on x86 and integrations test that will only compile for the target. So I prefer to run  `cargo test --lib` and  `cargo test --tests` with the respective `--target <...>`

I'll keep this task open, until I've added this info to the README and/or changed the example

t-moe, 2024-05-31:
Closing this, as this is now updated in the example repo and also added to the FAQ in the wiki

closed 2024-05-31 by t-moe