probe-rs/embedded-test #28: Logs are quite noisy

MabezDev, 2024-03-30:
```
running 4 tests
test tests::test_symestric_transfer                      ... INFO  Test exited with () or Ok(..)
└─ embedded_test::export::check_outcome @ /home/mabez/.cargo/git/checkouts/embedded-test-f175ea47b6323482/b67dec3/src/fmt.rs:37  
ok
test tests::test_asymestric_transfer                     ... INFO  Running test: Test { name: "tests::test_asymestric_transfer", function: 0x420011cc, should_panic: false, ignored: false, timeout: None }
└─ embedded_test::export::run_tests @ /home/mabez/.cargo/git/checkouts/embedded-test-f175ea47b6323482/b67dec3/src/fmt.rs:37  
INFO  Test exited with () or Ok(..)
└─ embedded_test::export::check_outcome @ /home/mabez/.cargo/git/checkouts/embedded-test-f175ea47b6323482/b67dec3/src/fmt.rs:37  
ok
test tests::test_symestric_transfer_huge_buffer          ... INFO  Test exited with () or Ok(..)
└─ embedded_test::export::check_outcome @ /home/mabez/.cargo/git/checkouts/embedded-test-f175ea47b6323482/b67dec3/src/fmt.rs:37  
ok
test tests::test_symestric_transfer_huge_buffer_no_alloc ... INFO  Test exited with () or Ok(..)
└─ embedded_test::export::check_outcome @ /home/mabez/.cargo/git/checkouts/embedded-test-f175ea47b6323482/b67dec3/src/fmt.rs:37  
ok
```

t-moe, 2024-03-30:
Try to remove the `defmt` (or `log`)  feature from the `embedded-test` crate.

I'll document the features asap. For now, refer to the Cargo.toml of the crate.

bugadani, 2024-04-05:
I think I'd rather if embedded-test collected logs into memory and printed them if a test failed. A log feature that interleaves itself into the "UI" output isn't ideal but keeping it enabled may still be useful.

t-moe, 2024-04-05:
There are two things here: 
1. I think @MabezDev was talking about the log messages from embedded-test (`Running Test...` and `Test Exited with...`). They can be disabled by removing the `log`/`defmt` feature. **The log/defmt output from the user will still be printed**
2. I think you @bugadani are talking about "output capturing". This is a different story. But Yes, we could implement that.

bugadani, 2024-04-06:
The issue opener is a bit vague on what Mabez wants, so yeah I guess I'm hijacking this a bit with output capture, but I feel like we can appropriate this issue for some brainstorming. I'm guessing capture would solve most of the clutter.

Anyway disregarding that, there's still one more aspect: embedded-test probably shouldn't print defmt location info by default, especially not for embedded-test-specific log strings. I'M not sure about the rest, I'm still not convinced that defmt should print any location info by default, but people seem to like that so `¯\_(ツ)_/¯`.