probe-rs/embedded-test #29: Rust analyzer does no longer show `Run Test | Debug` buttons

t-moe, 2024-04-04:
This is related to https://github.com/rust-lang/rust-analyzer/issues/17011 and can probably be fixed by emitting `#[::core::prelude::v1::test]` from my  macros. 

tokio test does this as well: https://github.com/tokio-rs/tokio/blob/a1acfd8c20475e8a878b012ac4aa90b361f004ae/tokio-macros/src/entry.rs#L365 )

Lets just hope that  `#[::core::prelude::v1::test]` stays a no-op, when `harness=false` for that particular test.

t-moe, 2024-12-03:
Simply emitting `#[::core::prelude::v1::test]` before the test function does not seem to work.  
Instead it looks like you have to manually convert the function into a token stream
https://github.com/tokio-rs/tokio/blob/master/tokio-macros/src/entry.rs#L589

couldnt get this to work in 30min. Not sure if needed.  keeping this issue open until I get more user feedback

jakobjung10, 2026-05-09:
Hi @t-moe,

#83 reenables the IDE support for rust-analyzer.
Is this in line with your expectations?

BR,
Jakob