probe-rs/embedded-test #31: defmt patches

jonathanpallant, 2024-05-30:
I'm aware there were some earlier defmt PRs that we didn't get as far as merging, so I wanted to come here and and say I'm very excited to see the project and its integration with probe-rs. I see this is currently depending on a git fork of defmt and I would love to talk about how we can get those upstream.

Yatekii, 2024-05-30:
cc @t-moe

t-moe, 2024-05-31:
Hi Jonathan,
Thanks for reaching out.

The just released version 0.4.0 of embedded-test, no longer depends on a fork of defmt.  
embedded-test can be used with defmt as a logging (& assertion) framework, but the users can also use the log crate and the assert macros from core.

Thanks again for the great inspiration you guys gave me with defmt-test!