probe-rs/embedded-test #35: Fail to run test: Unexpected semihosting command

xgroleau, 2024-07-12:
I'm having some issues using embedded test while trying to migrate away from defmt_test. It does seem to be an issue with probe-rs library not recognizing the semi-hosting commands so it might be more relevant to the probe-rs repo, but all other probe-rs functionalities works properly. 

This is using the jlink base as probe.

Here is all the relevant information
```
    Finished `test` profile [optimized + debuginfo] target(s) in 0.27s
     Running tests/example_test.rs (/home/xgroleau/Documents/wearable/cross/target/thumbv7em-none-eabihf/debug/deps/example_test-1c5c32917fc87a2a)
      Erasing ✔ [00:00:00] [#######################################################################################] 24.00 KiB/24.00 KiB @ 30.22 KiB/s (eta 0s )
  Programming ✔ [00:00:00] [#######################################################################################] 24.00 KiB/24.00 KiB @ 31.18 KiB/s (eta 0s )    Finished in 1.597s
Frame 0: syscall_readonly @ 0x00003ae6 inline
       /home/xgroleau/.local/share/cargo/registry/src/index.crates.io-6f17d22bba15001f/semihosting-0.1.11/src/sys/arm_compat/syscall/arm.rs:88:9
Frame 1: sys_exit_extended @ 0x0000000000003ae0 inline
       /home/xgroleau/.local/share/cargo/registry/src/index.crates.io-6f17d22bba15001f/semihosting-0.1.11/src/sys/arm_compat/mod.rs:196:9
Frame 2: exit @ 0x0000000000003ada inline
       /home/xgroleau/.local/share/cargo/registry/src/index.crates.io-6f17d22bba15001f/semihosting-0.1.11/src/sys/arm_compat/mod.rs:165:5
Frame 3: exit @ 0x0000000000003ada
       /home/xgroleau/.local/share/cargo/registry/src/index.crates.io-6f17d22bba15001f/semihosting-0.1.11/src/process.rs:14:5
Frame 4: panic_cold_display<i32> @ 0x00003b02
       /rustc/59e2c01c2217a01546222e4d9ff4e6695ee8a1db/library/core/src/panic.rs:98:1
Error: Unexpected semihosting command Unknown(UnknownCommandDetails { operation: 0, parameter: 537132968 }) cmdline_requested: true
error: test failed, to rerun pass `--test example_test`

Caused by:
  process didn't exit successfully: `probe-rs run --chip=nRF52840_xxAA --always-print-stacktrace /home/xgroleau/Documents/wearable/cross/target/thumbv7em-none-eabihf/debug/deps/example_test-1c5c32917fc87a2a` (exit status: 1)
```

The `test/example.rs`file
```rs
#![no_std]
#![no_main]

use nrf_softdevice as _; // for critical section impl
use defmt_rtt as _; // for logging

#[cfg(test)]
#[embedded_test::tests]
mod tests {

    #[test]
    fn example() {
        assert!(true)
    }
}

```

Cargo.toml file
```toml
[package]
name = "example_test"
version = "0.1.0"
edition = "2021"

[lib]
harness = false
test = false

[[test]]
name = "emedded-test-example"
harness = false

[dependencies]

[dev-dependencies]
embedded-test = { version = "0.4.0", features = ["defmt", "panic-handler"] }
nrf-softdevice = { version = "0.1.0", features = [ "nrf52840", "s140", "ble-peripheral", "critical-section-impl"]}
defmt-rtt = "0.4.1"
```

```sh
❯ probe-rs --version
probe-rs 0.24.0 (git commit: crates.io)
```

t-moe, 2024-07-12:
Thank you @xgroleau for reporting this.

Can you quickly try with probe-rs master?
This could be a duplicate of #34  which will be fixed as soon as the next probe-rs version is released.

xgroleau, 2024-07-12:
Thanks for the quick response @t-moe. Yes I confirm it works using master! It's probably the same issue since it looks very similar. I'll let you decide if you want to close this issue or wait for the release before closing this

closed 2024-12-16 by t-moe