probe-rs/embedded-test #37: ESP32-S3 issues

benedikt-bartscher, 2024-08-22:
Foremost, thank you for this awesome project. I am new to the rust embedded community and I really appreciate all the work.

I am currently trying to get tests running on an ESP32-S3-WROOM-1 (N16-R8) via the integrated usb jtag serial. Flashing and defmt logging via probe-rs works fine, but cargo test fails with the following error. If this is an issue with probe-rs instaed, please let me know, I'll move this issue if required.

cargo test
```log
   Compiling esp v0.1.0 (/home/bb/dev/*/esp)
    Finished `test` profile [optimized + debuginfo] target(s) in 0.83s
     Running tests/example.rs (target/xtensa-esp32s3-none-elf/debug/deps/example-b0a3b698ad8fbec5)
      Erasing ✔ [00:00:00] [##################################################################################################] 128.00 KiB/128.00 KiB @ 283.48 KiB/s (eta 0s )
  Programming ✔ [00:00:00] [#####################################################################################################] 26.91 KiB/26.91 KiB @ 64.72 KiB/s (eta 0s )    Finished in 0.88s
 WARN probe_rs::session: Could not clear all hardware breakpoints: An Xtensa specific error occurred.

Caused by:
    0: Xtensa debug module error
    1: Error writing register 0x47
    2: Register-specific error
 WARN probe_rs::session: Failed to deconfigure device during shutdown: An Xtensa specific error occurred.

Caused by:
    The result index of a batched command is not available.
Error: Failed to attach to RTT

Caused by:
    Error attempting to attach to RTT: Error communicating with probe: An Xtensa specific error occurred.
error: test failed, to rerun pass `--test example`

Caused by:
  proc
```

probe-rs --version
```
probe-rs 0.24.0 (git commit: 35dbf06)
```

Cargo.toml
```toml
...
embedded-test = { version = "0.4.0", features = [
  "panic-handler",
  "defmt",
  "embassy",
  "external-executor",
  "xtensa-semihosting",
] }
```

tests/example.rs
```rust
#![no_std]
#![no_main]

#[embedded_test::tests(executor = esp_hal::embassy::executor::thread::Executor::new())]
#[cfg(test)]
mod tests {
    use defmt_rtt as _;
    use esp_hal as _;

    #[init]
    fn init() -> () {
        defmt::info!("Init testing environment");
    }

    #[test]
    fn example() {
        assert!(true)
    }
}
```

t-moe, 2024-08-30:
Hei @benedikt-bartscher 
Thanks for reaching out.
I'll get back to this, after my vacation (3w). Thanks for your patience.

Yatekii, 2024-09-01:
Maybe @bugadani knows what's up here?

Seems to be a more general Xtensa issue rather than just `embedded-test`.

bugadani, 2024-09-01:
I have no idea which commit `35dbf06` is, embedded-test works just fine with latest probe-rs.

benedikt-bartscher, 2024-09-01:
@t-moe no worries, have a good one!

> Seems to be a more general Xtensa issue rather than just embedded-test.

@Yatekii it may be an issue with probe-rs itself, is there a simple way to test?

> I have no idea which commit `35dbf06` is, embedded-test on most ESP32 devices works just fine with latest probe-rs.

@bugadani I tried different versions, the output from above was accidentally from the version provided by the [AUR](https://aur.archlinux.org/packages/probe-rs). I am not sure why I can not find this commit sha in the probe-rs repo.

Latest master:
```bash
~> cargo install probe-rs-tools --git https://github.com/probe-rs/probe-rs --branch master
[...]
    Replaced package `probe-rs-tools v0.24.0 (https://github.com/probe-rs/probe-rs?tag=v0.24.0#6fc653ad)` with `probe-rs-tools v0.24.0 (https://github.com/probe-rs/probe-rs?branch=master#63c59d2f)` (executables `cargo-embed`, `cargo-flash`, `probe-rs`)

~> which probe-rs
/home/bb/.cargo/bin/probe-rs

~> probe-rs --version
probe-rs 0.24.0 (git commit: 63c59d2f)

~> cargo test
WARNING: use --release
  We *strongly* recommend using release profile when building esp-hal.
  The dev profile can potentially be one or more orders of magnitude
  slower than release, and may cause issues with timing-senstive
  peripherals and/or devices.

    Finished `test` profile [optimized + debuginfo] target(s) in 0.11s
     Running tests/example.rs (target/xtensa-esp32s3-none-elf/debug/deps/example-b6bee5eda2aa6e55)
      Erasing ✔ [00:00:00] [##################################################################################################] 128.00 KiB/128.00 KiB @ 224.42 KiB/s (eta 0s )
  Programming ✔ [00:00:01] [#####################################################################################################] 35.28 KiB/35.28 KiB @ 21.45 KiB/s (eta 0s )    Finished in 1.6459858s
 WARN probe_rs::session: Could not clear all hardware breakpoints: An Xtensa specific error occurred.

Caused by:
    0: Xtensa debug module error.
    1: Error reading register 0x45
    2: Register-specific error
 WARN probe_rs::session: Failed to deconfigure device during shutdown: Xtensa(XdmError(Xdm { narsel: 68, access: "reading", source: Error }))
Error: Error communicating with the probe.

Caused by:
    0: An Xtensa specific error occurred.
    1: The result index of a batched command is not available.
error: test failed, to rerun pass `--test example`

Caused by:
  process didn't exit successfully: `probe-rs run --chip esp32s3 /home/bb/dev/*/esp/target/xtensa-esp32s3-none-elf/debug/deps/example-b6bee5eda2aa6e55` (exit status: 1)
```

Tag v0.24.0:
```bash
~> cargo install probe-rs-tools --git https://github.com/probe-rs/probe-rs --tag v0.24.0 --locked
[...]
    Replaced package `probe-rs-tools v0.24.0 (https://github.com/probe-rs/probe-rs?branch=master#63c59d2f)` with `probe-rs-tools v0.24.0 (https://github.com/probe-rs/probe-rs?tag=v0.24.0#6fc653ad)` (executables `cargo-embed`, `cargo-flash`, `probe-rs`)

~> which probe-rs
/home/bb/.cargo/bin/probe-rs

~> probe-rs --version
probe-rs 0.24.0 (git commit: 6fc653ad)

~> cargo test
WARNING: use --release
  We *strongly* recommend using release profile when building esp-hal.
  The dev profile can potentially be one or more orders of magnitude
  slower than release, and may cause issues with timing-senstive
  peripherals and/or devices.

    Finished `test` profile [optimized + debuginfo] target(s) in 0.08s
     Running tests/example.rs (target/xtensa-esp32s3-none-elf/debug/deps/example-b6bee5eda2aa6e55)
      Erasing ✔ [00:00:00] [##################################################################################################] 128.00 KiB/128.00 KiB @ 296.36 KiB/s (eta 0s )
  Programming ✔ [00:00:00] [#####################################################################################################] 35.28 KiB/35.28 KiB @ 66.06 KiB/s (eta 0s )    Finished in 0.979s
 WARN probe_rs::session: Could not clear all hardware breakpoints: An Xtensa specific error occurred.

Caused by:
    0: Xtensa debug module error
    1: Error writing register 0x47
    2: Register-specific error
 WARN probe_rs::session: Failed to deconfigure device during shutdown: An Xtensa specific error occurred.

Caused by:
    The result index of a batched command is not available.
Error: Failed to attach to RTT

Caused by:
    Error attempting to attach to RTT: Error communicating with probe: An Xtensa specific error occurred.
error: test failed, to rerun pass `--test example`

Caused by:
  process didn't exit successfully: `probe-rs run --chip esp32s3 /home/bb/dev/*/esp/target/xtensa-esp32s3-none-elf/debug/deps/example-b6bee5eda2aa6e55` (exit status: 1)
```

I tried with `--release` as well - no difference.

bugadani, 2024-09-01:
Please paste the output of `espflash board-info`, too. I'll ask around if someone can reproduce this internally.

benedikt-bartscher, 2024-09-01:
Thank you very much!
```bash
~> espflash board-info
[2024-09-01T16:29:59Z INFO ] Serial port: '/dev/ttyACM0'
[2024-09-01T16:29:59Z INFO ] Connecting...
[2024-09-01T16:29:59Z INFO ] Using flash stub
Chip type:         esp32s3 (revision v0.2)
Crystal frequency: 40 MHz
Flash size:        16MB
Features:          WiFi, BLE
MAC address:       dc:da:0c:2a:29:a0
```

bugadani, 2024-09-04:
So far I can only tell that my reproducer works on my devkit :( Can you please verify that [this](https://github.com/bugadani/esp-embedded-test-async) fails for you? Just check out the repo and run `cargo +esp tt`.

benedikt-bartscher, 2024-09-04:
Thank's @bugadani for looking into this. I tried your example repo and it works just fine with all of our dev boards. While comparing our repo to yours i finally found the issue:

build.rs
```
# ours
println!("cargo:rustc-link-arg-bins=-Tlinkall.x");
# yours
println!("cargo::rustc-link-arg=-Tlinkall.x");
```

I accidentially only linked bins which does not apply to the tests... Such a stupid mistake. Sorry for bothering and thank's again for your help!

bugadani, 2024-09-04:
Ah, that's a good resolution to this issue :)

closed 2024-09-04 by MabezDev