probe-rs/embedded-test #40: Missing executor with feature embassy

KorribanMaster, 2024-10-01:
## Description
embedded_test fails to compile with the following error message
```
   Compiling embedded-test v0.4.0
error[E0432]: unresolved import `embassy_executor::Executor`
  --> /home/hechte/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embedded-test-0.4.0/src/export.rs:21:9
   |
21 | pub use embassy_executor::Executor; // Please activate the `executor-thread` or `executor-interrupt` feature on the embassy-executor crate (v0.5.x)!
   |         ^^^^^^^^^^^^^^^^^^^^^^^^^^ no `Executor` in the root
   |
help: consider importing this struct instead
   |
21 | pub use embassy_executor::raw::Executor; // Please activate the `executor-thread` or `executor-interrupt` feature on the embassy-executor crate (v0.5.x)!
   |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

For more information about this error, try `rustc --explain E0432`.
error: could not compile `embedded-test` (lib) due to 1 previous error
```
Although I have enabled an executor in my Cargo.toml
Here is the relevant snippets from the cargo.toml
```
[package]
edition = "2021"
name = "pn"
version = "0.1.0"
license = "MIT OR Apache-2.0"

[[bin]]
name = "pn"
test = false
bench = false

[lib]
harness = false

[dependencies]
embassy-stm32 = { version = "0.1.0", features = [ "defmt", "time-driver-any", "stm32g474rc", "memory-x", "unstable-pac", "exti"]  }
embassy-sync = { version = "0.6.0", features = ["defmt"] }
embassy-executor = { version = "0.6.0", features = ["arch-cortex-m", "executor-thread", "defmt", "integrated-timers"] }
embassy-time = { version = "0.3.2", features = ["defmt", "defmt-timestamp-uptime", "tick-hz-32_768"] }
embassy-usb = { version = "0.3.0", features = ["defmt"] }
embassy-futures = { version = "0.1.0"}
usbd-hid = "0.8.1"

defmt = "0.3"
defmt-rtt = "0.4"

cortex-m = { version = "0.7.6", features = ["critical-section-single-core"] }
cortex-m-rt = "0.7.0"
embedded-hal = "1.0.0"
embedded-can = { version = "0.4" }
panic-probe = { version = "0.3", features = ["print-defmt"] }
heapless = { version = "0.8", default-features = false }
static_cell = "2.0.0"
embedded-test = { version = "0.4.0", features = ["embassy", "defmt"] }

[dev-dependencies]
defmt-test = "0.3"
embedded-test = { version = "0.4.0", features = ["embassy", "defmt"] }

# cargo build/run
[profile.dev]
codegen-units = 1
debug = 2
debug-assertions = true # <-
incremental = false
opt-level = 'z'         # <-
overflow-checks = true  # <-

# cargo test
[profile.test]
codegen-units = 1
debug = 2
debug-assertions = true # <-
incremental = false
opt-level = 3           # <-
overflow-checks = true  # <-

# cargo build/run --release
[profile.release]
codegen-units = 1
debug = 2
debug-assertions = false # <-
incremental = false
lto = 'fat'
opt-level = 3            # <-
overflow-checks = false  # <-

# cargo test --release
[profile.bench]
codegen-units = 1
debug = 2
debug-assertions = false # <-
incremental = false
lto = 'fat'
opt-level = 3            # <-
overflow-checks = false  # <-

[patch.crates-io]
embassy-stm32 = { git = "https://github.com/embassy-rs/embassy", rev="7648d42b7f23a2caad29ed6e16123b088ccdc8b5" }
embassy-sync = { git = "https://github.com/embassy-rs/embassy", rev="7648d42b7f23a2caad29ed6e16123b088ccdc8b5" }
embassy-executor = { git = "https://github.com/embassy-rs/embassy", rev="7648d42b7f23a2caad29ed6e16123b088ccdc8b5" }
embassy-time = { git = "https://github.com/embassy-rs/embassy", rev="7648d42b7f23a2caad29ed6e16123b088ccdc8b5" }
embassy-usb = { git = "https://github.com/embassy-rs/embassy", rev="7648d42b7f23a2caad29ed6e16123b088ccdc8b5" }
```

Can anyone explain why embedded_test does not pickup the executor_thread feature?

t-moe, 2024-10-01:
Please try with the master branch. 
I havent made a release since merging https://github.com/probe-rs/embedded-test/pull/38

KorribanMaster, 2024-10-01:
Hey thanks for your super fast response i patched embedded-test but had no luck with it either. i updated my issue above accordingly

t-moe, 2024-10-01:
this is strange.
can you quickly also share the output of `cargo tree`  to validate we're using the same embassy-executor crate in the entire tree?

KorribanMaster, 2024-10-01:
Will close this issue

KorribanMaster, 2024-10-02:
Hey unfortunately after a reorganization of the code the issue came back so it seems I have not yet found the underlying cause.
Thus I have made a small demo repository which can be used to replicate the issue
https://github.com/KorribanMaster/embedded-test-issue-40
I have used cargo tree in this repository as you have suggested and i can see that embedded-test tries to bring along it's own version of embassy-executor (v0.5 instead of v0.6)
```
❯ cargo tree
self-tests v0.1.0 (/home/hechte/git/embedded-test-issue#40)
├── cortex-m v0.7.7
│   ├── bare-metal v0.2.5
│   │   [build-dependencies]
│   │   └── rustc_version v0.2.3
│   │       └── semver v0.9.0
│   │           └── semver-parser v0.7.0
│   ├── bitfield v0.13.2
│   ├── critical-section v1.1.3
│   ├── embedded-hal v0.2.7
│   │   ├── nb v0.1.3
│   │   │   └── nb v1.1.0
│   │   └── void v1.0.2
│   └── volatile-register v0.2.2
│       └── vcell v0.1.3
├── cortex-m-rt v0.7.3
│   └── cortex-m-rt-macros v0.7.0 (proc-macro)
│       ├── proc-macro2 v1.0.86
│       │   └── unicode-ident v1.0.13
│       ├── quote v1.0.37
│       │   └── proc-macro2 v1.0.86 (*)
│       └── syn v1.0.109
│           ├── proc-macro2 v1.0.86 (*)
│           ├── quote v1.0.37 (*)
│           └── unicode-ident v1.0.13
├── defmt v0.3.8
│   ├── bitflags v1.3.2
│   └── defmt-macros v0.3.9 (proc-macro)
│       ├── defmt-parser v0.3.4
│       │   └── thiserror v1.0.64
│       │       └── thiserror-impl v1.0.64 (proc-macro)
│       │           ├── proc-macro2 v1.0.86 (*)
│       │           ├── quote v1.0.37 (*)
│       │           └── syn v2.0.79
│       │               ├── proc-macro2 v1.0.86 (*)
│       │               ├── quote v1.0.37 (*)
│       │               └── unicode-ident v1.0.13
│       ├── proc-macro-error v1.0.4
│       │   ├── proc-macro-error-attr v1.0.4 (proc-macro)
│       │   │   ├── proc-macro2 v1.0.86 (*)
│       │   │   └── quote v1.0.37 (*)
│       │   │   [build-dependencies]
│       │   │   └── version_check v0.9.5
│       │   ├── proc-macro2 v1.0.86 (*)
│       │   ├── quote v1.0.37 (*)
│       │   └── syn v1.0.109 (*)
│       │   [build-dependencies]
│       │   └── version_check v0.9.5
│       ├── proc-macro2 v1.0.86 (*)
│       ├── quote v1.0.37 (*)
│       └── syn v2.0.79 (*)
├── defmt-rtt v0.4.1
│   ├── critical-section v1.1.3
│   └── defmt v0.3.8 (*)
├── defmt-test v0.3.2
│   ├── cortex-m-rt v0.7.3 (*)
│   ├── cortex-m-semihosting v0.5.0
│   │   └── cortex-m v0.7.7 (*)
│   ├── defmt v0.3.8 (*)
│   └── defmt-test-macros v0.3.1 (proc-macro)
│       ├── proc-macro2 v1.0.86 (*)
│       ├── quote v1.0.37 (*)
│       └── syn v2.0.79 (*)
├── embassy-executor v0.6.0 (https://github.com/embassy-rs/embassy?rev=7648d42b7f23a2caad29ed6e16123b088ccdc8b5#7648d42b)
[... 193 lines cut ...]
│   │   ├── proc-macro2 v1.0.86 (*)
│   │   ├── quote v1.0.37 (*)
│   │   └── syn v2.0.79 (*)
│   ├── heapless v0.8.0 (*)
│   ├── semihosting v0.1.15
│   ├── serde v1.0.210 (*)
│   └── serde-json-core v0.5.1
│       ├── heapless v0.7.17
│       │   ├── hash32 v0.2.1
│       │   │   └── byteorder v1.5.0
│       │   └── stable_deref_trait v1.2.0
│       │   [build-dependencies]
│       │   └── rustc_version v0.4.1
│       │       └── semver v1.0.23
│       ├── ryu v1.0.18
│       └── serde v1.0.210 (*)
├── heapless v0.8.0 (*)
├── panic-probe v0.3.2
│   ├── cortex-m v0.7.7 (*)
│   └── defmt v0.3.8 (*)
├── static_cell v2.1.0
│   └── portable-atomic v1.9.0
└── usbd-hid v0.8.2 (*)

```

t-moe, 2024-10-02:
thanks for the reproducer. I'll tackle this soon.

KorribanMaster, 2024-10-07:
I have forked the repo and set embassy executor to v0.6 only and that solved the issue. Apparently setting a range for allowed external crates comes with a lot of issues in rust. The resolver might pick two different versions depending on your cached contents and many other variables so it is not advisable to rely on the resolver. In this specific case my suggestion would be to use feature flags like `embassy-05` and `embassy-06` to allow an explicit selection of the executor.

t-moe, 2024-10-07:
sorry for the delay on my side (I'm a bit swamped with my day-job). But I'm glad you found a solution that works. I'll keep this issue open, until its fixed in this repo and released.

KorribanMaster, 2024-10-18:
Thanks for keeping me in the loop. I can live with my solution for a while so take your time

omani, 2024-10-20:
@KorribanMaster your fork does **not** fix it for me.

KorribanMaster, 2024-10-20:
Ah, you might have to move embedded-test to dependencies. At least for me, this sometimes changed things while I was investigating

omani, 2024-10-20:
tried that too. doesn't work.

omani, 2024-10-20:
so I guess right now `embedded-test` is unusable with embassy. are you guys planning to fix this? @t-moe what is the status of this issue?

omani, 2024-10-20:
ah now I get it:
https://github.com/embassy-rs/embassy/blob/c9358e1f1e5d88aa0ad998c44eb2da6be73cf477/embassy-executor/src/lib.rs#L38C1-L38C64

one has to enable the arch feature. in my case (for an esp32c3) I need **arch-riscv32**:
```
embassy-executor    = { version = "=0.5.0",  features = ["arch-riscv32", "executor-thread", "defmt", "integrated-timers"] }
```

but notice how this only works with `embassy-executor` version `0.5.0`. as soon as I switch to version `0.6.0` (which I need), it breaks again with the same error.

omani, 2024-10-20:
considering your comment:
> I have forked the repo and set embassy executor to v0.6 only and that solved the issue. Apparently setting a range for allowed external crates comes with a lot of issues in rust. The resolver might [...]

using your fork, with fixed version `0.6.0` doesn't work either.

so far, Im only able to use embassy-executor 0.5.0. and I dont quite understand what the difference here is (embassy-executor hasn't changed from 0.5.0 to 0.6.0 in this context). maybe it has something to do with the cache like you said.

omani, 2024-10-20:
ok investigating further I made it work with version 0.6.0.
@KorribanMaster you are funny man. why dont you tell us that your fork is not different from upstream?

**ONE HAS TO PIN THE COMMIT REV IN CARGO.TOML** to get your commit (fix 0.6.0 version in Cargo.toml):
```
embedded-test = { git = "https://github.com/KorribanMaster/embedded-test.git", rev = "3157a7f74ce886acbc751d7b505a6c651284efdc", features = ["defmt", "embassy"] }
```

now everything works.

omani, 2024-10-20:
conclusion:
a version range does not work because rust is doing weird things then.
a fixed version works.

@t-moe I support @KorribanMaster's suggestion about different feature flags for different `embassy-executor` versions. see https://github.com/probe-rs/embedded-test/issues/40#issuecomment-2396077605

KorribanMaster, 2024-10-20:
Oh I guess I never thought someone else would use my fork. However, I always pin everything. I patch to a  commit sha as this ensures stability and is generally considered good practice

omani, 2024-10-20:
> Oh I guess I never thought someone else would use my fork. However, I always pin everything. I patch to a commit sha as this ensures stability and is generally considered good practice

yeah Im just kidding. my bad for not checking out the commit :)

omani, 2024-10-20:
everything works and Im happy now. I let you guys decide what you want to do with this issue.

thanks everyone.

t-moe, 2024-10-21:
Released to crates.io as 0.5.0.
Hope this fixes your issues. Otherwise just comment again.

closed 2024-10-21 by t-moe

t-moe, 2024-11-06:
> .... I would like to use the git version instead of the crates.io version if possible....

When you override embassy-executor with a git dependency, `cargo tree | grep embassy-executor` will show two different versions being used . The fix is to add a patch section and keep the original embassy-executor dependency.

```toml
[dependencies]
embassy-executor = { version = "0.6.2", features = [ "arch-cortex-m", "executor-thread", "executor-interrupt", "defmt", "integrated-timers", ] }

[patch.crates.io]
embassy-executor = { git = "https://github.com/embassy-rs/embassy" }

```

tullom, 2024-11-06:
Thank you!!! that fixed the issue. 😄