probe-rs/embedded-test #41: Check if esp partition-table is honored when using embedded-test

t-moe, 2024-10-17:


akauppi, 2024-11-05:
I would like to know more about this. Could it be connected to #43 I just filed?

t-moe, 2024-11-06:
(@akauppi  I've added some details in the issue, but i dont think this is related to your issue  #43 )

Regarding this issue here: I've validated and can confirm that probe-rs will happily discard at least my previous ota_data partition.

@bugadani  I've quickly checked the probe-rs source and cannot see that the test run mode does anything special regarding idf partitions. Do you have any ideas, why my partitions could get lost when using probe-rs? I dont assume you erase the entire flash, instead you seem to be using espflash internally, all pretty standard....

bugadani, 2024-11-06:
probe-rs will happily overwrite your bootloader and partition table with a default one, unless you specify them with `--idf-partition-table` and `--idf-bootloader`

bugadani, 2024-11-06:
Even better, even if you specify your partition table, I don't think probe-rs overwrites ota_data and you might end up programming one of the OTA partitions, then the bootloader may still boot the other one. espflash has code around it, but I'm not sure probe-rs does.

bugadani, 2024-11-06:
All right, I need some more info then. Can you flash your test firmware, and then connect espflash and paste the bootloader's output?

t-moe, 2024-11-06:
> Even better, even if you specify your partition table, I don't think probe-rs overwrites ota_data and you might end up programming one of the OTA partitions, then the bootloader may still boot the other one. espflash has code around it, but I'm not sure probe-rs does.

Yeah, I was afraid, that probe-rs will probably not erase/update ota_data (and I didnt know espflash can do it either), but thats not what I observed...

I will try to create a reproducer

bugadani, 2024-11-06:
Regardless, this isn't an embedded-test issue but a probe-rs one

t-moe, 2024-11-07:
Ok, here is the reproducer for the esp32c6.

partitions.csv
```csv
# Name,    Type, SubType, Offset,  Size, Flags
ota_data,  data, ota,     0x9000,   0x2000, # 8KB for OTA state (must be exactly 8KB)
prod_conf, 0x40, 0,       0xB000,   0x2000, # 8KB (2 pages) for production configuration
ota_0,     app,  ota_0,   0x20000,  1M,
ota_1,     app,  ota_1,   0x120000, 1M,
```

1. flash the merged binary ([merged.bin.zip](https://github.com/user-attachments/files/17658866/merged.bin.zip) ), that will boot from the 2nd ota partition: `espflash write-bin 0x00 merged.bin`  and also contain a production configuration partition.

2. validate the targets boots from the 2nd ota partition by running `espflash monitor` and pressing CTRL+R:

> I (48) boot: Partition Table:
> I (51) boot: ## Label            Usage          Type ST Offset   Length
> I (58) boot:  0 ota_data         OTA data         01 00 00009000 00002000
> I (66) boot:  1 prod_conf        unknown          40 00 0000b000 00002000
> I (73) boot:  2 ota_0            OTA app          00 10 00020000 00100000
> I (81) boot:  3 ota_1            OTA app          00 11 00120000 00100000
> I (88) boot: End of partition table
> I (93) esp_image: segment 0: paddr=00120020 vaddr=42000020 size=28954h (166228) map
> I (135) esp_image: segment 1: paddr=0014897c vaddr=40800000 size=00014h (    20) load
> I (136) esp_image: segment 2: paddr=00148998 vaddr=42028998 size=0436ch ( 17260) map
> I (144) esp_image: segment 3: paddr=0014cd0c vaddr=40800014 size=00f6ch (  3948) load
> I (150) boot: **Loaded app from partition at offset 0x120000**
> I (155) boot: Disabling RNG early entropy source...
> INFO - **Hello From Partition 2!**

3. Validate our secret is still there `espflash read-flash 0xB000 100 prod_conf_out.txt && cat prod_conf_out.txt`:
> <....>
> IMPORTANT_SECRET

4. Flash a hello world ([hello_world.elf.zip](https://github.com/user-attachments/files/17658652/hello_world.elf.zip)) with probe-rs `probe-rs run hello_world.elf --idf-partition-table partitions.csv`

5. Validate output of `espflash monitor` again:
> I (48) boot: Partition Table:
> I (51) boot: ## Label            Usage          Type ST Offset   Length
> I (58) boot:  0 ota_data         OTA data         01 00 00009000 00002000
> I (66) boot:  1 prod_conf        unknown          40 00 0000b000 00002000
> I (73) boot:  2 ota_0            OTA app          00 10 00020000 00100000
> I (81) boot:  3 ota_1            OTA app          00 11 00120000 00100000
> I (88) boot: End of partition table
> I (93) boot: **No factory image, trying OTA 0**
> I (97) esp_image: segment 0: paddr=00020020 vaddr=42000020 size=06314h ( 25364) map
> I (111) esp_image: segment 1: paddr=0002633c vaddr=40800000 size=00014h (    20) load
> I (114) esp_image: segment 2: paddr=00026358 vaddr=42006358 size=021a0h (  8608) map
> I (125) esp_image: segment 3: paddr=00028500 vaddr=40800014 size=00f2ch (  3884) load
> I (132) boot: **Loaded app from partition at offset 0x20000**
> I (154) boot: **Set actual ota_seq=1 in otadata[0]**
> I (155) boot: Disabling RNG early entropy source...
> INFO - **Hello From probe-rs!**

6. Our secret is gone `espflash read-flash 0xB000 100 prod_conf_out.txt && cat prod_conf_out.txt`
> ...������������....

This shows that probe-rs will overwrite our prod_conf.bin. It will also overwrite ota_data, which would actually be legitimate in that particular scenario, but AFAIK neither probe-rs nor espflash have code to handle ota_data, so I assume ota_data is just overwritten as well.