probe-rs/embedded-test #43: Unable to run tests on ESP32-C6: "The target did not respond with test list until timeout."

akauppi, 2024-11-05:
**Steps**

```
$ git clone https://github.com/probe-rs/embedded-test-example
$ cd embedded-test-example
$ cargo test
[...]
    Finished `test` profile [unoptimized + debuginfo] target(s) in 0.18s
     Running tests/example_test.rs (/home/ubuntu/target/riscv32imac-unknown-none-elf/debug/deps/example_test-ca528a9059f152d1)
      Erasing ✔ 100% [####################] 320.00 KiB @ 107.63 KiB/s (took 3s)
  Programming ✔ 100% [####################]  87.98 KiB @   2.43 KiB/s (took 36s)                                                                                                                    Finished in 36.14s
Frame 0: <unknown function @ 0x40017606> @ 0x40017606
Frame 1: <unknown function @ 0x400233e8> @ 0x400233e8
Error: The target did not respond with test list until timeout.
error: test failed, to rerun pass `--test example_test`
```

My system:

```
$ lsb_release -a
[...]
Description:	Ubuntu 24.04.1 LTS

$ probe-rs --version
probe-rs 0.24.0 (git commit: 3421e932)

$ rustc --version
rustc 1.82.0 (f6e511eec 2024-10-15)
```

The `probe-rs` is installed from Git, but I expect that should not be a problem?   Noticed that the `embedded-test-example` lacks somewhat behind in its dependencies, e.g. `esp-hal` 0.17.0 instead of 0.21.1.

How to get the example repo to run the tests?

akauppi, 2024-11-05:
+ `embedded-test-example` is still using `embedded-test` 0.4.0 (not 0.5.0)

t-moe, 2024-11-06:
Were you using a ota partitions before flashing embedded-test? (Probe-rs and also espflash will not automatically erase the ota-data partition...)
If yes, try erasing the chip first....

I'll check the example later today, when I'm back at the computer

t-moe, 2024-11-06:
I've tried to reproduce the error, by using a custom partition-table, without success. I dont think your issue is related to #41 .

Can you share more details?  e.g. rerun `cargo test` with `RUST_LOG=info`

Do you use a custom partition table? (or was the relation to #41 just a hunch?)

Also: can you try with an erased chip?

bugadani, 2024-11-06:
This issue happens because the chip is stuck in download mode. Power cycling or `espflash monitor` and crtl-r to reboot usually brings it out of that state, although I guess probe-rs should be able to do it as well, we just don't.

akauppi, 2024-11-09:
Thanks for the advice. I'll remember the Ctrl-R but didn't need it this time.. I was three days away, and now things suddenly work. Closing.

closed 2024-11-09 by akauppi