probe-rs/embedded-test #47: Replace `rtt-log` crate

t-moe, 2024-11-15:
Sourcebox via Matrix:
> I've seen `embedded-test` uses my `rtt-log` crate. While I appreciate this, I'm not sure if it's really a good idea in the long-term because of possible version conflicts with `rtt-target`. Because my crate as well as `embedded-test` has it as dependency, the version must match. The functionality of `rtt-log` is very little and there was some discussion long ago to directly include logging into `rtt-target`. Someone just has to do it.

Also: rtt-log does not allow to logging in blocking rtt mode (only discard on overflow)

sourcebox, 2024-11-16:
If it's desired to have logging in a separate crate, I could also transfer ownership of `rtt-log` to the `probe-rs` organisation. That way, it would be easier to keep the versions in sync and add new features.

t-moe, 2024-12-03:
Ok, so rtt-target 0.6.1 has been released. 

Furthermore, the next embedded-test version will no longer depend on rtt-target or rtt-log, instead the users are encouraged to use something like `#[embedded_test::tests(setup=rtt_target::rtt_init_log!())]` if they want to setup logging early.

closed 2024-12-03 by t-moe