probe-rs/embedded-test #48: Struggling to compile for STM32 + Embassy

Gerharddc, 2024-11-18:
Hi, I am having some problems trying to compile for an STM32 device using the Embassy feature. Specifically I am getting the following error:

```
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/usr/local/rustup/toolchains/1.82.0-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/usr/local/rustup/toolchains/1.82.0-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/usr/local/rustup/toolchains/1.82.0-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/vscode/vscode-server/bin/linux-x64/f1a4fb101478ce6ec82fe9627c43efbf9e98c813/bin/remote-cli:/usr/local/cargo/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/renode" VSLANG="1033" "rust-lld" "-flavor" "gnu" "/tmp/rustcKeC7Xx/symbols.o" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/example_test-506bc196bb296178.example_test.3adfa79f423e9ba7-cgu.0.rcgu.o" "--as-needed" "-Bstatic" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libembedded_test-2f47b171bd446a78.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libserde_json_core-f7b102a896219ee5.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libryu-9b8ac7fa030860b8.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libheapless-d82220867c9f45a1.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libhash32-bc3fde7fc0766731.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libembassy_executor-820bc2b2b7a88e36.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libembassy_time_queue_driver-e0b3eaec1736e32f.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libembassy_time_driver-da26f9d365ccf7c4.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libcortex_m-41b95f1de0e491bf.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libembedded_hal-096cd4a01692998b.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libvoid-04e456fdcefa2ca7.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libnb-9c673c85dc70e794.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libnb-401912cda3822599.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libvolatile_register-150dd79df8f724b6.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libvcell-2068e77239f39278.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libbare_metal-39e2e98538789012.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libsemihosting-8542fd8f31f3390a.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libheapless-71aafbca25202d85.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libstable_deref_trait-c4c3713d6a70d1e0.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libserde-fc4218c1e85f5222.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libhash32-dd47769c86729aef.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libbyteorder-2a94d01e5f2337fd.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libdefmt_rtt-b3403a9743b6dc3f.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libcritical_section-c11e0f0b9a158325.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libdefmt-18a4b6fcd58a710d.rlib" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libbitflags-b1fefe807fa9b5f3.rlib" "/usr/local/rustup/toolchains/1.82.0-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/librustc_std_workspace_core-b84a106e6eb97d6d.rlib" "/usr/local/rustup/toolchains/1.82.0-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcore-af18ab7ca8da5674.rlib" "/usr/local/rustup/toolchains/1.82.0-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/libcompiler_builtins-5315fb2b7d32c1bb.rlib" "-Bdynamic" "--eh-frame-hdr" "-z" "noexecstack" "-L" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/build/mcu-fw-app-a541cbb169d23e2a/out" "-L" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/build/cortex-m-9173d4d8ee2c5dd1/out" "-L" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-04fcf0f2b8e66895/out" "-L" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/build/defmt-265ce0b466760fbe/out" "-L" "/usr/local/cargo/git/checkouts/stm32-data-generated-cb34dad5f3150296/3aa0a2a/stm32-metapac/src/chips/stm32f439zi" "-L" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/build/embedded-test-78fe3d4ceb093108/out" "-o" "/workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/example_test-506bc196bb296178" "--gc-sections" "-O1" "--nmagic" "-Tlink.x" "-Tdefmt.x" "-Tembedded-test.x"
  = note: rust-lld: error: /workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-04fcf0f2b8e66895/out/link.x:51: symbol not found: DefaultHandler_
          rust-lld: error: undefined symbol: _embassy_time_now
          >>> referenced by lib.rs:182 (src/lib.rs:182)
          >>>               embassy_time_driver-da26f9d365ccf7c4.embassy_time_driver.da2bbffc02910678-cgu.0.rcgu.o:(.text._ZN19embassy_time_driver3now17h2e0ad9c5d9dbba66E+0x0) in archive /workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libembassy_time_driver-da26f9d365ccf7c4.rlib
          
          rust-lld: error: undefined symbol: _embassy_time_allocate_alarm
          >>> referenced by lib.rs:189 (src/lib.rs:189)
          >>>               embassy_time_driver-da26f9d365ccf7c4.embassy_time_driver.da2bbffc02910678-cgu.0.rcgu.o:(.text._ZN19embassy_time_driver14allocate_alarm17hb4e5373be07bab3fE+0x0) in archive /workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libembassy_time_driver-da26f9d365ccf7c4.rlib
          
          rust-lld: error: undefined symbol: _embassy_time_set_alarm_callback
          >>> referenced by lib.rs:194 (src/lib.rs:194)
          >>>               embassy_time_driver-da26f9d365ccf7c4.embassy_time_driver.da2bbffc02910678-cgu.0.rcgu.o:(.text._ZN19embassy_time_driver18set_alarm_callback17h04a234ab406e87afE+0x0) in archive /workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libembassy_time_driver-da26f9d365ccf7c4.rlib
          
          rust-lld: error: undefined symbol: _embassy_time_set_alarm
          >>> referenced by lib.rs:199 (src/lib.rs:199)
          >>>               embassy_time_driver-da26f9d365ccf7c4.embassy_time_driver.da2bbffc02910678-cgu.0.rcgu.o:(.text._ZN19embassy_time_driver9set_alarm17h768a5f1590cd36dbE+0x0) in archive /workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/libembassy_time_driver-da26f9d365ccf7c4.rlib
          
          rust-lld: error: /workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-04fcf0f2b8e66895/out/link.x:51: symbol not found: DefaultHandler_
          rust-lld: error: 
          BUG(cortex-m-rt): the reset vector is missing
          
          rust-lld: error: 
          BUG(cortex-m-rt): the exception vectors are missing
          
          rust-lld: error: 
          ERROR(cortex-m-rt): The interrupt vectors are missing.
          Possible solutions, from most likely to less likely:
          - Link to a svd2rust generated device crate
          - Check that you actually use the device/hal/bsp crate in your code
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
          may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.
          
          rust-lld: error: /workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-04fcf0f2b8e66895/out/link.x:51: symbol not found: DefaultHandler_
          rust-lld: error: 
          BUG(cortex-m-rt): the reset vector is missing
          
          rust-lld: error: 
          BUG(cortex-m-rt): the exception vectors are missing
          
          rust-lld: error: 
          ERROR(cortex-m-rt): The interrupt vectors are missing.
          Possible solutions, from most likely to less likely:
          - Link to a svd2rust generated device crate
          - Check that you actually use the device/hal/bsp crate in your code
          - Disable the 'device' feature of cortex-m-rt to build a generic application (a dependency
          may be enabling it)
          - Supply the interrupt handlers yourself. Check the documentation for details.
```

[... cut ...]
Finally, I believe the `cortex-m-rt` linking issues are because of my dependency on `cortex-m-rt` and it's linker script through `println!("cargo:rustc-link-arg=-Tlink.x");`. I believe this is necessary for things to actually run on the STM32 though. With this library, one usually needs to explicitly set the entry point using `cortex_m_rt::entry`, but I believe that the `#[embassy_executor::main]` macro automatically configures that under normal conditions. However, I have no idea what the equivalent would be when using embedded_test instead of Embassy directly.

Any ideas or suggestions would be greatly appreciated, thanks!

t-moe, 2024-11-18:
Hi @Gerharddc ,
I'm currently on mobile, but here some tipps:
* use `cargo tree` to ensure only a single embassy-executor is selected accross your entire dependency tree.
* try to bring at least one function of your hal into scope, to ensure the exception handlers are linked. See here: https://github.com/probe-rs/embedded-test-example/blob/23ab16d6a8bd60e984d1b207cf50f1769ad8e023/tests/example_test.rs#L9 (example is a bit outdated, but the concept still holds)

Gerharddc, 2024-11-18:
@t-moe thanks, that was very useful! I have now managed to get things to compile by updating my example_test.rs to the following:

```
#![no_std]
#![no_main]

#[cfg(test)]
#[embedded_test::tests(executor = embassy_executor::Executor::new())]
mod tests {
    use defmt_rtt as _;
    use embassy_stm32::time::Hertz;
    use embassy_stm32::Config;
    use embassy_stm32::Peripherals;

    #[init]
    async fn init() -> Peripherals {
        let mut config = Config::default();
        {
            use embassy_stm32::rcc::*;
            config.rcc.hse = Some(Hse {
                freq: Hertz(8_000_000),
                mode: HseMode::Bypass,
            });
            config.rcc.pll_src = PllSource::HSE;
            config.rcc.pll = Some(Pll {
                prediv: PllPreDiv::DIV4,
                mul: PllMul::MUL168,
                divp: Some(PllPDiv::DIV2), // 8mhz / 4 * 168 / 2 = 168Mhz.
                divq: Some(PllQDiv::DIV7), // 8mhz / 4 * 168 / 7 = 48Mhz.
                divr: None,
            });
            config.rcc.ahb_pre = AHBPrescaler::DIV1;
            config.rcc.apb1_pre = APBPrescaler::DIV4;
            config.rcc.apb2_pre = APBPrescaler::DIV2;
            config.rcc.sys = Sysclk::PLL1_P;
        }
        embassy_stm32::init(config)
    }

    #[test]
    async fn it_works(_state: Peripherals) {
        let result = 2 + 2;
        assert_eq!(result, 4);
    }
}
```

It now also flashes and runs but it seems probe-rs now crashes with the following:

```
Frame 0: syscall_readonly @ 0x080050a0 inline
       /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/semihosting-0.1.16/src/sys/arm_compat/syscall/arm.rs:88:9
Frame 1: sys_exit_extended @ 0x000000000800509c inline
       /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/semihosting-0.1.16/src/sys/arm_compat/mod.rs:196:9
Frame 2: exit @ 0x0000000008005096 inline
       /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/semihosting-0.1.16/src/sys/arm_compat/mod.rs:165:5
Frame 3: exit @ 0x0000000008005096
       /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/semihosting-0.1.16/src/process.rs:57:5
Frame 4: sys_get_cmdline @ 0x080050c0
       /usr/local/cargo/registry/src/index.crates.io-6f17d22bba15001f/semihosting-0.1.16/src/sys/arm_compat/mod.rs:216:1
Error: Unexpected semihosting command Unknown(UnknownCommandDetails { operation: 0, parameter: 537066388 }) cmdline_requested: true
error: test failed, to rerun pass `--test example_test`

Caused by:
  process didn't exit successfully: `probe-rs run --chip STM32F439ZITx /workspaces/mcu-fw/target/thumbv7em-none-eabihf/debug/deps/example_test-0f37cfa56785a4ae` (exit status: 1)
```

Could this be a limitation of the ST-Link that I am using as a programmer/debugger?

t-moe, 2024-11-18:
Are you using probe-rs master?

See https://github.com/probe-rs/embedded-test/issues/34

Gerharddc, 2024-11-18:
@t-moe no I was using the latest stable release. Switching to master fixed the problem so now everything works perfectly. Thanks!