probe-rs/embedded-test #49: Embedded-test cannot reset MIMXRT685-EVK by itself and when manually resetting, it intermittently fails

tullom, 2024-11-18:
I am evaluating embedded-test using an [NXP MIMXRT685](https://www.nxp.com/design/design-center/development-boards-and-designs/i-mx-evaluation-and-development-boards/i-mx-rt600-evaluation-kit:MIMXRT685-EVK) board, however the board is failing to reset itself:

```
running 2 tests
test tests::test_example       ...  WARN probe_rs::vendor::nxp::sequences::nxp_armv8m: FlexSPI0 NOR flash config block starts with 0x00000000 (valid blocks start with 0x42464346)
Frame 0: <unknown function @ 0x0001c04a> @ 0x0001c04a
Frame 1: <unknown function @ 0x1300cba4> @ 0x1300cba4
FAILED
test tests::test_example_panic ...  WARN probe_rs::vendor::nxp::sequences::nxp_armv8m: FlexSPI0 NOR flash config block starts with 0x00000000 (valid blocks start with 0x42464346)
Frame 0: <unknown function @ 0x0001c04a> @ 0x0001c04a
Frame 1: <unknown function @ 0x1300cba4> @ 0x1300cba4
FAILED

failures:

---- tests::test_example ----
Test timed out after 60s

---- tests::test_example_panic ----
Test timed out after 60s
```

In addition, if I manually reset the chip by pressing the onboard reset button, I still get an error 99% of the time

```
running 2 tests
test tests::test_example       ...  WARN probe_rs::vendor::nxp::sequences::nxp_armv8m: FlexSPI0 NOR flash config block starts with 0x00000000 (valid blocks start with 0x42464346)
Error: An ARM specific error occurred.

Caused by:
    0: Error using access port FullyQualifiedApAddress { dp: Default, ap: V1(0) }.
    1: Failed to read register DRW at address 0x0c
    2: An error occurred in the communication with an access port or debug port.
    3: Target device responded with a FAULT response to the request.
error: test failed, to rerun pass `--test example_test`
```

Once or twice, I've seen it work as intended when pressing the reset button. This leads me to believe that it might be a timing issue.

By chance, I enabled RUST_LOG = "debug" and many debug lines from probe-rs are written to the console. After flashing and manually pressing the reset button, the tests end up running **successfully 100% of the time**. (!!!) I am just speculating, but I think that due to the added delay with console IO, the NXP chip finishes its debug port startup sequence (in hardware) when it comes out of reset. I'm not sure where to insert these delays to fix my issue (is this an embedded-test problem or a probe-rs problem?)

repo to reproduce here: [tullom/mimxrt685-eval](https://github.com/tullom/mimxrt685-eval)
simply comment and uncomment `RUST_LOG = "debug"` to see the difference

t-moe, 2024-11-19:
This is probably not related to embededded-test, but to probe-rs flashing/resetting. Does probe-rs work for you when not using embedded-test?

tullom, 2024-11-19:
Indeed probe-rs works on both the flashing side and the resetting side. For example, the chip is able to flash, reset itself, and run flashed binaries with `cargo run`.

bugadani, 2024-11-19:
I'm not familiar with these chips unfortunately.