probe-rs/embedded-test #50: Importing logic from current crate

Owez, 2024-11-20:
Hi,

There's not currently an example of how to test the logic inside of the crate, e.g., `use example::hi; assert_eq!(hi(), true)` where `example` is the crate the `tests/` directory is in. I've been having some issues trying to do this locally in my project, and using embassy examples plus embedded-test on top. I generally get:

```
error[E0432]: unresolved import `stm32f0`
 --> tests/my_test.rs:7:9
  |
7 |     use stm32f0::hi;
  |         ^^^^^^^ use of undeclared crate or module `stm32f0`
```

If I try different variations and placements of imports (plus trying extern crate). Is importing code from the crate supported, and if not what is the way to actually hook into logic? I feel like I just can't get any enviroments set up properly and this is user error on my part 😄

Thanks

t-moe, 2024-11-20:
hi @Owez ,
embedded-test adds nothing special in that regard.
You have to add a lib target to your crate and then you can refer to the crate by name from the integration test.

I'm currently on mobile this week, so I cant provide a complete example. But you I'm sure you can find help in the embedded rust matrix channel or probe-rs channel.

Owez, 2024-11-20:
Gotcha, I'll have a go now – I didn't know there was a matrix. I'll close this issue because it's a mistake on my end

closed 2024-11-20 by Owez