probe-rs/embedded-test #59: How to spawn a task with embassy?

romainreignier, 2025-04-03:
Hi,
Thanks for this nice tool.
I would like to init an embassy-net stack to communicate with another device during the tests to check the communication.
But embassy-net needs a network task to be spawned but I don't see how to retrieve the Spawner to spawn a new task.
Do you think that this is something that can be added?

t-moe, 2025-04-04:
Hi @romainreignier ,

* You could maybe try using [Spawner::for_current_executor](https://docs.embassy.dev/embassy-executor/git/std/struct.Spawner.html#method.for_current_executor) ? 

* If this does not work, I would take a PR that passes the spawner to the method passed via `embedded_test::tests(setup=func)`.

* Or maybe #15 would also help?

romainreignier, 2025-04-04:
Hi @t-moe 

Thanks a lot, the [Spawner::for_current_executor](https://docs.embassy.dev/embassy-executor/git/std/struct.Spawner.html#method.for_current_executor) call works, I did not know this function.
For now, I have just tried a simple log task but it should work for the network related tasks.

closed 2025-04-04 by romainreignier