probe-rs/embedded-test #62: Unit-Test support (opposed to integration-tests only)

t-moe, 2025-06-03:
I have some ideas on how to tackle this already, but it needs some changes on the host-side as well and is a bigger endeavor

t-moe, 2025-06-18:
Work in progress
https://github.com/probe-rs/embedded-test/tree/next
https://github.com/probe-rs/probe-rs/tree/embedded_test_v2

t-moe, 2025-06-19:
Making good progress here.

What works so far:
* [x] Run **unit tests** (lib) on esp32c6 (riscv). Probe-rs reads them from ELF directly. 
* [x] Run **unit tests** (lib) on std with embedded-test-std-runner. Tests are collected with linkme crate and reported via json as before.

Things left to do:
* [ ] Test whether my [ELF Symbols](https://github.com/probe-rs/embedded-test/blob/next/macros/src/attributes/tests/codegen/export_sym.rs#L60)  can be read by probe-rs on arm and xtensa. 
When I exported this symbol on x86, my test case would issue a SIGSEV in the dynamic loader :open_mouth:  
Is it ok, if I use your CI/CD cluster again @bugadani  ?
* [ ] Test whether bin unit tests work
* [ ] Cleanup

bugadani, 2025-06-20:
> Is it ok, if I use your CI/CD cluster again

of course, please do

t-moe, 2025-08-07:
I've just released embedded test [0.7.0-alpha.0](https://crates.io/crates/embedded-test/0.7.0-alpha.0) which supports unit tests!

* Needs probe-rs master.
* Does not support ariel os right now, because of [src/export.rs#L28-L34](https://github.com/probe-rs/embedded-test/blob/next/src/export.rs#L28-L34) @kaspar030 please advise.

Working branch is still [next](https://github.com/probe-rs/embedded-test/tree/next), but it will soon replace master.

t-moe, 2025-08-20:
0.7.0-alpha.2 is out, with embassy-executor 0.8

t-moe, 2025-08-29:
0.7.0-alpha.3 is out, with embassy-executor 0.9

@kaspar030 Were you able to test one of the 0.7.0-alpha.* versions ?

kaspar030, 2025-08-29:
> Were you able to test one of the 0.7.0-alpha.* versions ?

No, unfortunately we're lagging behind with updating our dependencies. I've tried hacking in just the necessary updates (embassy-executor to 0.9.0 and embedded-test to 0.7.0-alpha), but the embassy timer crates somehow bind it all together, there's too much else to update for a quick test.

kaspar030, 2025-08-29:
> * Does not support ariel os right now, because of [src/export.rs#L28-L34](https://github.com/probe-rs/embedded-test/blob/next/src/export.rs?rgh-link-date=2025-08-07T09%3A42%3A22.000Z#L28-L34) [@kaspar030](https://github.com/kaspar030) please advise.

It would be totally fine releasing with untested Ariel support. We're currently pinning an older version in the build system that transcends to all Ariel applications. Once we caught up with the dependencies, we can do testing/fixing/bumping embedded-test.