probe-rs/embedded-test #70: Panic handler setup with 0.7.0-alpha

kevswims, 2025-08-25:
I am trying to get unit and integration setup using the 0.7.0-alpha.2 and I seem to be chasing an issue with panic handling in circles. I have an existing embassy project that has been using defmt and embassy on an STM32 device. 

I have created a basic example_test.rs file in a tests directory that I would like to run but I cannot get it to compile. I have tried several different configurations with regards to the panic handlers but I cannot find a configuration setup that will compile.

With this `embedded-test` entry in my Cargo.toml

```
embedded-test = { version = "0.7.0-alpha.2", default-features = false, features = ["defmt", "embassy", "semihosting", "panic-handler"]}
```

I get the following compilation error:

```
error[E0152]: duplicate lang item in crate `panic_probe` (which `firmware` depends on): `panic_impl`
  |
  = note: the lang item is first defined in crate `embedded_test` (which `example_test` depends on)
  = note: first definition in `embedded_test` loaded from /home/kevin/source/luci/luci-safety/firmware-rs/target/thumbv7em-none-eabihf/debug/deps/libembedded_test-ed74a13d712a5545.rlib
  = note: second definition in `panic_probe` loaded from /home/kevin/source/luci/luci-safety/firmware-rs/target/thumbv7em-none-eabihf/debug/deps/libpanic_probe-813730bb554a1a2b.rlib, /home/kevin/source/luci/luci-safety/firmware-rs/target/thumbv7em-none-eabihf/debug/deps/libpanic_probe-813730bb554a1a2b.rmeta
```

I then tried turning off the `panic-handler` feature for embedded-test which results in this error:

```
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/kevin/.rustup/toolchains/1.85-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/usr/lib/ccache:/home/kevin/gems/bin:/home/kevin/.local/bin:/home/kevin/source/yocto-kirkstone/sources/meta-luci/.env/bin:/usr/lib/ccache:/home/kevin/gems/bin:/home/kevin/.nvm/versions/node/v20.15.0/bin:/home/kevin/.cargo/bin:/home/kevin/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin:/usr/local/go/bin:/home/kevin/.vscode/extensions/ms-python.debugpy-2025.10.0-linux-x64/bundled/scripts/noConfigScripts:/home/kevin/.config/Code/User/globalStorage/github.copilot-chat/debugCommand:/usr/local/go/bin" VSLANG="1033" "rust-lld" "-flavor" "gnu" "/tmp/rustcN7GWEU/symbols.o" "<1 object files omitted>" "--as-needed" "-Bstatic" "/home/kevin/source/luci/luci-safety/firmware-rs/target/thumbv7em-none-eabihf/debug/deps/{libfirmware-1db68f4d910c4c60.rlib,libthiserror-a402b86b123181fb.rlib,libassign_resources-35180df41d7701be.rlib,libnom-8b6be24fdf2f9afe.rlib,libmemchr-ea97da60b4936d40.rlib,libpanic_probe-813730bb554a1a2b.rlib,libdefmt_rtt-b8481765b11febe0.rlib,libbincode-0560a341691add29.rlib,libunty-eadbe96e0173ae64.rlib,libpid-1bf6a5158d53c56e.rlib,libembassy_stm32-b8e368d477dfd917.rlib,libembedded_hal_nb-1df202a3bf055522.rlib,libblock_device_driver-165f835583759162.rlib,libaligned-537b53185cbee8ab.rlib,libas_slice-3f7bbb3a6151d27f.rlib,librand_core-60db20e8ad9ae633.rlib,librand_core-c9a8a108b954996a.rlib,libstm32_fmc-b0eb871e15ef77a0.rlib,libstatic_assertions-4e9159a395b2d962.rlib,libembassy_usb_synopsys_otg-143c9b6fcd36a35f.rlib,libsdio_host-335f681eaa8f6614.rlib,libchrono-745ebe685b35aa10.rlib,libembassy_embedded_hal-c88c77356caa6d8e.rlib,libembedded_storage_async-4dac14b13ee026a7.rlib,libembedded_storage-c7e609526a808075.rlib,libfutures_util-5544b9e23880d6be.rlib,libpin_project_lite-87757761396501f1.rlib,libfutures_task-fa5600125f381297.rlib,libpin_utils-32e89c873980688a.rlib,libbit_field-26da4d257813bb56.rlib,libembedded_can-89eff0d5f8d9e89c.rlib,libcfg_if-742c34dffa953b23.rlib,libembassy_time_queue_utils-4f37df010333d492.rlib,libstm32_metapac-14e5375eb65286ce.rlib,libcortex_m_rt-ec6e0fe6079ad8fe.rlib,libembassy_hal_internal-81b8c4959f68e876.rlib,libnum_traits-28585736e7e9ad5f.rlib,libfirmware_rs_icd-e5d67d9efa6e111f.rlib,libpostcard_rpc-e13c6d65425a7228.rlib,libembassy_usb-f727201330654294.rlib,libusbd_hid-7d07177e3cc6e635.rlib,libusb_device-fc55a3847ed9718e.rlib,libssmarshal-a90d23ab03adc155.rlib,libencode_unicode-8c1219929008f70f.rlib,libembassy_net_driver_channel-866cc01220751a1f.rlib,libembassy_net_driver-7f14ca465e9544e2.rlib,libstatic_cell-6a461e84d0195fbd.rlib,libportable_atomic-028f2e060f222c0e.rlib,libembassy_usb_driver-46e0c08b810c3701.rlib,libembassy_time-e9c1c525eb6a3d94.rlib,libembedded_hal_async-481f62934aa26329.rlib,libembedded_hal-d554603d79c380ff.rlib,libembassy_time_driver-b33fe325a75ae831.rlib,libembassy_sync-b1b4d98c645f902c.rlib,libfutures_sink-99461e6eb7e499b2.rlib,libembedded_io_async-89d7c53b969b22b1.rlib,libembedded_io-b5d4a3d1780ae538.rlib,libfutures_core-bfe5907e66c6fb1e.rlib,libembassy_futures-11d558b42c5f4bb5.rlib,libpostcard-a762755198283d58.rlib,libheapless-676f0f8ec6478c46.rlib,libhash32-de85a8a198f64928.rlib,libcobs-208ab0f4bbffa622.rlib,libserde_big_array-54b2439f97139231.rlib,libbytemuck-92828240a533584f.rlib,libpostcard_schema-952673e527fadbdb.rlib,libheapless-7d20b8720dcdb792.rlib,libstable_deref_trait-fb1cdc610c868206.rlib,libdefmt-4078ba4562af715f.rlib,libhash32-99460988fc96d6a6.rlib,libbyteorder-6b1504be7b486592.rlib,libserde-24112b965fcab073.rlib,libembedded_test-66e398205d329380.rlib,libembassy_executor-343cb61d784e371d.rlib,libdefmt-656de94b4a0ca6b8.rlib,libbitflags-bf4e724ed7e91919.rlib,libembassy_executor_timer_queue-c276b82d07a51f6b.rlib,libcortex_m-4e132dd2c63764f3.rlib,libembedded_hal-f4ad2d9d044ce77f.rlib,libvoid-44615c66d22c46e3.rlib,libnb-b59240d0acc79ca3.rlib,libnb-09f31a239c34d170.rlib,libcritical_section-3125efedcaf578c6.rlib,libvolatile_register-ac1947c9a982ba89.rlib,libvcell-56f7a1d526f33750.rlib,libbare_metal-170cb040c0f11d3d.rlib,libsemihosting-64e57e5f58f6048d.rlib}" "/home/kevin/.rustup/toolchains/1.85-x86_64-unknown-linux-gnu/lib/rustlib/thumbv7em-none-eabihf/lib/{librustc_std_workspace_core-89717189c5caa977.rlib,libcore-e6122f30404945e0.rlib,libcompiler_builtins-5581ff22e90261f7.rlib}" "-Bdynamic" "--eh-frame-hdr" "-z" "noexecstack" "-L" "/home/kevin/source/luci/luci-safety/firmware-rs/target/thumbv7em-none-eabihf/debug/build/firmware-426beacb490e68f5/out" "-L" "/home/kevin/source/luci/luci-safety/firmware-rs/target/thumbv7em-none-eabihf/debug/build/cortex-m-875443d189929af8/out" "-L" "/home/kevin/source/luci/luci-safety/firmware-rs/target/thumbv7em-none-eabihf/debug/build/cortex-m-rt-b1b59ff4255bb8f4/out" "-L" "/home/kevin/source/luci/luci-safety/firmware-rs/target/thumbv7em-none-eabihf/debug/build/defmt-c414ef10a8bdc24e/out" "-L" "/home/kevin/.cargo/git/checkouts/stm32-data-generated-a5ed3f1859eb8323/6f80b25/stm32-metapac/src/chips/stm32h7s3l8" "-L" "/home/kevin/source/luci/luci-safety/firmware-rs/target/thumbv7em-none-eabihf/debug/build/embedded-test-linker-script-fcb1c1137cfcfeac/out" "-o" "/home/kevin/source/luci/luci-safety/firmware-rs/target/thumbv7em-none-eabihf/debug/deps/example_test-0780d54900a4b6d1" "--gc-sections" "-O1" "-Tembedded-test.x"
  = note: some arguments are omitted. use `--verbose` to show all linker arguments
  = note: rust-lld: error: undefined symbol: _defmt_panic
          >>> referenced by mod.rs:133 (/home/kevin/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/defmt-1.0.1/src/export/mod.rs:133)
          >>>               /home/kevin/source/luci/luci-safety/firmware-rs/target/thumbv7em-none-eabihf/debug/deps/example_test-0780d54900a4b6d1.example_test.2d07ac5102b49f8b-cgu.0.rcgu.o:(embassy_executor::arch::thread::Executor::run::h415e996493b1f109)
          >>> referenced by mod.rs:133 (/home/kevin/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/defmt-1.0.1/src/export/mod.rs:133)
          >>>               embassy_stm32-b8e368d477dfd917.embassy_stm32.cd20cf7f5458e794-cgu.0.rcgu.o:(embassy_stm32::rcc::_version::init_pll::h6aab6650bd185a74) in archive /home/kevin/source/luci/luci-safety/firmware-rs/target/thumbv7em-none-eabihf/debug/deps/libembassy_stm32-b8e368d477dfd917.rlib
          >>> referenced by mod.rs:133 (/home/kevin/.cargo/registry/src/index.crates.io-1949cf8c6b5b557f/defmt-1.0.1/src/export/mod.rs:133)
          >>>               embassy_stm32-b8e368d477dfd917.embassy_stm32.cd20cf7f5458e794-cgu.0.rcgu.o:(embassy_stm32::rcc::_version::init_pll::h6aab6650bd185a74) in archive /home/kevin/source/luci/luci-safety/firmware-rs/target/thumbv7em-none-eabihf/debug/deps/libembassy_stm32-b8e368d477dfd917.rlib
          >>> referenced 24 more times
```

Is there a way to setup embedded-test in a project that is already using panic-probe? Am I missing something else in my configuration here?

kevswims, 2025-08-25:
Ahh, that was a good tip, I missed that one of our lib files that that `use panic_probe as _;` statement in it. 

With that removed though I am still getting a linker error for `undefined symbol: _defmt_panic` even with the `panic-handler` feature enabled for embedded-test.

t-moe, 2025-08-25:
> I added this to my test file and it compiles now. I'm not sure if this is the correct way around this though?
> 
> ```
> #[defmt::panic_handler]
> fn defmt_panic() -> ! {
>     loop {}
> }
> ```

No, this is not helpful, as we need semihosting exit instead of busy looping.
Does the linker error get away when you add a `use defmt as _;` to the testfile?

kevswims, 2025-08-25:
I just discovered all my rustc-link-args in my build.rs had the `-bins` on them:

```
    println!("cargo:rustc-link-arg-bins=--nmagic");
    println!("cargo:rustc-link-arg-bins=-Tlink.x");
    println!("cargo:rustc-link-arg-bins=-Tdefmt.x");
```

I changed these to this without the `-bins` and now it builds. 
```
    println!("cargo:rustc-link-arg=--nmagic");
    println!("cargo:rustc-link-arg=-Tlink.x");
    println!("cargo:rustc-link-arg=-Tdefmt.x");
    println!("cargo:rustc-link-arg=-Tembedded-test.x");
```

I am now getting this error:
```
0.110636 [ERROR] Querying tests via semihosting is no longer supported by embedded-test. Please upgrade probe-rs to the latest version (embedded_test embedded-test-0.7.0-alpha.2/src/fmt.rs:62)
Error: Unexpected semihosting command ExitError(ExitErrorDetails { reason: 131110, exit_status: Some(134), subcode: None }) cmdline_requested: true
error: test failed, to rerun pass `--test example_test`
```

I am running this on an STM32H7S3 which has a separate bootloader on it so I'm not sure if that is the issue or something else?

My `probe-rs` version is `0.29.1`.

kevswims, 2025-08-25:
Awesome, that got things running. Thanks for the help! 

Tomorrow I'll try writing some real tests and see what else I can get going.